BruceDevices/firmware #1385: Latest 1.10.2 missing features mentioned in release notes

emericklaw, 2025-06-11:
**Describe the bug:**
The 1.10.2 release is missing at least these changes mentioned in the release notes:

    WebUI - Fixed Drag 'n drop not keeping the actual folder/fs after upload
    Enhancement for webInterface https://github.com/pr3y/Bruce/pull/1342 by @lshaf
    Web UI tweaks https://github.com/pr3y/Bruce/pull/1370 by @emericklaw
    Tweak webui for save and run https://github.com/pr3y/Bruce/pull/1378 by @lshaf 

I have not had time to go through the other changes to see if they are also missing.

bmorcelli, 2025-06-11:
these things weren't added in the 1.10,2
they are in the dev brench for 1.11

bmorcelli, 2025-06-11:
I need to talk to you about the WebUI> Device navigation, I have some ideas that we can put in motion with minor changes

do you have Discord or something? Send me your nickname or add me there