BruceDevices/firmware #1413: Sub-GHZ sanity check?

JimikoSK, 2025-06-19:
Everything seems to be working except Sub-GHZ. I had to reseat the IPEX connector the other day.

Before I buy an external antenna to mod, any sanity check, icon, whatever that shows Bruce is recognizing it?

I tried listening to a small Sub-GHZ FM remote, but didn't pick anything up, so just wanted to see if there is a sanity check I can try. 😅

JimikoSK, 2025-06-19:
@bmorcelli 
> Device? Module?

Oh sorry, T-Embed CC1101.

bmorcelli, 2025-06-19:
* Remove SdCard
* Config> Factory Reset
* RF> Config>RF Module > M5 Module, it will change to the single pinned module attached to 43/44

After that, back to :
* RF> Config> RF Module> CC1101 (if it shows CC1101 not found, you have broken the CC chip somehow, if not, CC1101 is working, and you might have an antenna problem

JimikoSK, 2025-06-19:
> * Remove SdCard
> * Config> Factory Reset
> * RF> Config>RF Module > M5 Module, it will change to the single pinned module attached to 43/44
>
> After that, back to :
>
> * RF> Config> RF Module> CC1101 (if [...]

@bmorcelli Thanks! Let me give it a try. 💕

closed 2025-06-19 by bmorcelli