BruceDevices/firmware #1414: Iram overflow error while compiling bruce

discobeast, 2025-06-19:
**Describe the bug:**
Compiler raising error due to "iram0_0_seg overflow"

**Steps To Reproduce:**
platformio.exe 'run', '--environment', 'Marauder-v7'"
- Device used: trying to compile bruce for marauder-v7

**Expected behavior:**
Bruce compiles

**Actual behavior:**
Platformio raises overflow error

**Additional context / logs:**
Linking .pio\build\Marauder-v7\firmware.elf
Memory region         Used Size  Region Size  %age Used
     iram0_0_seg:      138120 B       128 KB    105.38%
     iram0_2_seg:     2755627 B    3342304 B     82.45%
c:/users/discobeast/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/Marauder-v7/firmware.elf section `.iram0.text' will not fit in region `iram0_0_seg'
     dram0_0_seg:      102420 B     124580 B     82.21%
     drom0_0_seg:      702518 B    4194272 B     16.75%
c:/users/discobeast/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
    rtc_iram_seg:         103 B       8176 B      1.26%
c:/users/discobeast/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 7048 bytes
    rtc_data_seg:         108 B       8176 B      1.32%
collect2.exe: error: ld returned 1 exit status
    rtc_slow_seg:          24 B       7680 B      0.31%
  extern_ram_seg:          0 GB         4 MB      0.00%
*** [.pio\build\Marauder-v7\firmware.elf] Error 1
================================================================================================================================= [FAILED] Took 227.06 seconds =================================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
Marauder-v7    FAILED    00:03:47.064
============================================================================================================================= 1 failed, 0 succeeded in 00:03:47.064 =============================================================================================================================

 *  The terminal process "C:\Users\Discobeast\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'Marauder-v7'" terminated with exit code: 1.

bmorcelli, 2025-06-19:
You are not using bruce framework, probably

discobeast, 2025-06-19:
Wdym?

discobeast, 2025-06-19:
I cloned the repo and trying to build with platformio..
Idk what other framework there would be

bmorcelli, 2025-06-19:
[image]

It is working normally.. follow the instructions and try again

https://github.com/pr3y/Bruce/wiki/Building-from-source

discobeast, 2025-06-20:
I rebuilt the entire workspace and even ran another on a github remote repo and it still didnt work :\
Idk wth is wrong

closed 2025-06-20 by discobeast