BruceDevices/firmware #1427: Bruce not working on CoreS3 SE since 1.9.1

emericklaw, 2025-06-24:
**Describe the bug:**
When flashing a 1.10.x version of the firmware it does not boot. I get the following on the serial port:

```
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (614) esp_ota_ops: not found otadata
ESP-ROM:esp32s3-20210327
```

I went back through the firmware versions and 1.9.1 was the last version to work.

**Steps To Reproduce:**

Flash firmware newer than 1.9.1 and it will get stuck in a boot loop.

- Device used: M5 CoreS3 SE

**Expected behavior:**

Firmware should boot.

**Actual behavior:**

Boot loop.

**Additional context / logs:**

bmorcelli, 2025-08-01:
Interesting... I am finishing a few things over other device (Lilygo T-LoraPager) and will look towards this device 

Probably it is conflicting these GROVE_SDA and GROVE_SCL with the internal I2C control Bus..

The SDCard issue I need to study a little more.. probably changes on the upstream library messed with it.. I will check it out too

cyba3r, 2025-08-01:
Great! Thank you a lot! I tested during the last hours a couple of things, changing espressif version and so on. But with no success. I removed the code from "setupSdCard" with the working code from M5Stack (https://docs.m5stack.com/en/arduino/m5cores3/sdcard). Just insert:

 SPI.begin(SD_SPI_SCK_PIN, SD_SPI_MISO_PIN, SD_SPI_MOSI_PIN, SD_SPI_CS_PIN);

    if (!SD.begin(SD_SPI_CS_PIN, SPI, 25000000)) {.....

But also without success. I will play now a bit around with "GROVE_SDA"....

cyba3r, 2025-08-01:
When I change GROVE_SDA  to 12 and  GROVE_SCL to 11 (default I2c on CoreS3) then the device doesn't go into boot loop :-) But the SD still struggling....

cyba3r, 2025-08-02:
Ok, I got it to work now. 1 out of 10 Cards working now. Most of my 8GB aren't working, but 1 of my 64GB cards working!  For my GPS Modul (https://docs.m5stack.com/en/module/Module%20GPS%20v2.0) I had to add the following: 	

-DGPS_SERIAL_TX=17
-DGPS_SERIAL_RX=18

Then this also works perfectly. I will adding now GPS information when using packet sniffing :-) 

Thanks for your great work on this project!!

bmorcelli, 2025-08-02:
Try different values for the SPI frequency here

https://github.com/pr3y/Bruce/blob/main/boards%2Fm5stack-cores3%2Fpins_arduino.h#L101

Change to 40000000 and 80000000