BruceDevices/firmware #1429: T-watch S3 BadBLE crash

CronosCtrl, 2025-06-25:
Everytime I try to setup BadBLE on the lilygo watch, it crashes and reboots Bruce. 

After selecting the keyboard layout language, it just crashes and reboots. 

Is there a fix or a way around it possible?

CronosCtrl, 2025-06-26:
> It doesn't only happen on the Lilygo watch, I use my M5StickC Plus2 and on the 2nd time I setup, it crashes right after I choose the keyboard layout.
> So I think there's an error, and it comes from the null pointer (trash pointer), I recommend fix like this:
> ```cpp
> if (hid == nullptr) {
>      hid = new HIDInterface();
>      if (!hid || hid == nullptr) {
>            displayError("HID is null", true);
>            returnToMenu = true;
>            goto EXIT;
>      }
> }
> ```
> before `ducky_chooseKb(hid, ble);`
> I tried it on the second time and so on, it worked!
> **How it works**:
> - By checking if the `hid` variable is null (`nullptr`), create a new `HIDInterface` class
> - Check again if after create there's an error, it will display the error and exit the loop.
> 
> If I am wrong, I apologize my mistake.

O ok. Do you think it will be added on next update? Do you have a fork where I could use your code mod?

bmorcelli, 2025-06-26:
the problem is that there's a missplaced flag in this device, so it is behaving as if it has Keyboard, like T-Deck or Cardputer.

i will fix it here and push on Beta asap