BruceDevices/firmware #1433: Failed to mount sd card after CC1101 installed

Ohminecraft, 2025-06-27:
I have My DIY Hardware i mod the original lilygo t-embed non cc1101 to work with my diy hardware and it works without another extension except sd card, now i connect with my cc1101 i restart device and sd card not work :(
how can i fix this

pins_arduino.h: https://gist.github.com/Ohminecraft/f5a6d78daca831fbf61c4113dc65cbeb
interface.cpp: https://gist.github.com/Ohminecraft/cecdc7a2c96e1779047e6dbcaa0f22d7
lilygo-t-embed.ini: https://gist.github.com/Ohminecraft/5c438a66a94452551b766da03c085f05

Huzzla101, 2025-06-28:
This dont work on all devices but u can try .... When you using 2 modules
at same time the SD card reader tends to have issues working well ... So
one solution that may work sometimes is to use the same GND for booth
modules. Just attach the sd card GND to same pin as cc1101 antenna GND
uses... And maybe it will work... Lets hope ..

Ohminecraft, 2025-06-30:
> This dont work on all devices but u can try .... When you using 2 modules
> at same time the SD card reader tends to have issues working well ... So
> one solution that may work sometimes is to use the [...]

Thank you for reply i finally fixed sd card thank you for supporting me i will close this issue

closed 2025-06-30 by Ohminecraft