BruceDevices/firmware #1454: IR T-Watch s3

RossFallESP32, 2025-07-03:
Before this update, I flashed via Bruce computer web, which at that time did not have the main firmware. Previously, I used the beta version, IR worked fine, but the latest version, IR does not work, the LED does not show the work. How should I fix it? But I am not sure that when I press the boot button, it may have touched the LED because it is close to each other. So I consulted the code first. If the code is not a problem, I may have to check the circuit. Thank you for all the advice in advance.

Huzzla101, 2025-07-03:
I would suggest that you ,before engaging into the circuit etc, to try and
re flash the Software again , go for the beta version that is current if ur
using the default version now, then if beta have the exact problem i would
try last resort to flash to default version again (as it may have been
updated now since last time you did flash it) . If your lucky thats all
that is needed..

Huzzla101, 2025-07-03:
> Thanks bro, I think so too, but I'm not sure because the complete firmware
> is just released in this version, so I'm asking for advice to chase the
> problems in order, and I'm studying T-Watch s3 in detail for future use,
> and new versions of the library change some usage instructions, so I chase
> the problems one by one, starting from Jake's code.

Oohh thats cool.. !  =)   I dont have a t-watch s3 yet but since Bruce is
there i might as well get me one.... So  i dont have the needed "expertise"
to serve you more on that field but it seems like you know what your doing
so it will go smoothly... Off course you have checked the config meny in IR
to double check pins is the same as last time it worked etc (if there are
such meny on T-watch) .... Are you also on the Discord server that is
related to Bruce Enhancement tool? Would be cool to have you update the
people from all over the world about the knowledge you will obtain from
studying that watch as it looks very interesting indeed..Also the chat
have other T watch users that may help.. Im sure you do alot of cool stuff
besides that to... Its an interesting world we live in for sure....