BruceDevices/firmware #1455: Black screen on display after a long time having installed Bruce T-Embed

datyosu, 2025-07-03:
My new T-Embed arrived about 3 weeks ago (I waited 2 months for shipping), and i installed Bruce correctly.
Today, like any other day I was going to use it, it stopped working as usual. The screen got stuck at startup. I used the RST button, and it turned off and on again, but with a black screen. The screen has no problems at all; it turns on correctly, just with a black display.
I tried formatting it and completely erasing the firmware with Esptool and reinstalling Bruce, but the same problem persisted.
I tried installing the factory firmware, and the same thing persisted: the screen and LEDs were on, but the display was black.
I don't know what else to do. I spent 3 hours researching the same problem and only found two people, one on Reddit and one here on the troubleshooting forum. Neither solved the problem.
I can't afford to buy another T-Embed. Pls help help help.

datyosu, 2025-07-03:
@Huzzla101 I thought exactly the same thing moments after calming down, the battery from 2 hours ago is already disconnected, I want to leave it long enough for it to discharge or something to happen, idk.
The esp32 seems to be working fine, but I've noticed that the speaker or some of the software is damaged, I've come to think that the memory was damaged or corrupted in the worst case.
I will find out if there is any command to verify or test the operation of the screen, if you know something like that please tell me. help help help

datyosu, 2025-07-06:
I tried exactly what you said, bro.
Unfortunately, I still can't find a solution.
I disconnected the Flex cable (I don't know what it's called in your country), but it's the cable that connects the screen to the ESP32. It's still giving me the same problem. All the devices, like the C1101 and the PN532, are disconnected from the ESP32, and the same problem persists.
I'd like to think it's a manufacturing issue.
I've spoken with an acquaintance on Bruce's Discord server, and he told me he also has a T-Embed with the same problem, using the Beta firmware, and that since then, the screen turns on and also controls the brightness, but still has no image.
I'll try to find a solution or a way to identify the problem.
In the meantime, I'll try to contact Lilygo support for a helpful answer or solution.
Help Help Help

JudgeLP, 2025-07-17:
I am having exactly the same problem after one week using normaly my T-embed cc1101 with Bruce 1.10.2

I wrote to Lilygo Oficial Store from AliExpress and i am waiting fir a solution.

JudgeLP, 2025-07-18:
Following Lilygo official Store i proceed to flash lates factory firmare: 

could you please flash our firmware again to check if it can work normally ? https://github.com/Xinyuan-LilyGO/T-Embed-CC1101/tree/master/firmware

And it seems that screen is working again. Next days i will see if things keep working or not before trying again latest Bruce firmware, and i will inform.

https://github.com/user-attachments/assets/b2f87b96-1ed5-4c8a-8ce9-c78e1363b12d

HiennNek, 2025-08-27:
@datyosu Hi, I'm sorry for this late reply, but I believe the information I have might be helpful to you. 

I'm experiencing the same issue you described (like, 100% exactly), which typically occurs for me after using the device for about a week. After some testing, I suspect that the SPI line of the ESP32-S3 Wroom 1U may be malfunctioning, causing the display to show a black screen with no colors. However, since the backlight LED of the screen is connected to a different GPIO, that part should still function.

To confirm that we are experiencing the same problem, could you please use a multimeter to measure the voltage on GPIO pins 9, 10, and 11 of the MCU while running the code below? If any of them show a voltage lower than 3.3V, then it seems we have a similar issue.

Also, I have tried to replace the screen; as you would expect, it didn't work, and/but the original T-Embed screen still works after plugging it into another ESP32.

Btw, are you Vietnamese? If so, we could communicate in Vietnamese.

Code:
```cpp
void setup() {
  pinMode(9, OUTPUT);
  pinMode(10, OUTPUT);
  pinMode(11, OUTPUT);
}

void loop() {
  digitalWrite(9, HIGH);
  digitalWrite(10, HIGH);
  digitalWrite(11, HIGH);
}
```
Image:

[image]

HiennNek, 2025-09-02:
@datyosu I sent an email to Lilygo, but they haven't responded for a week. Their customer service is poor :(

datyosu, 2025-09-02:
@HiennNek 
I would believe that asking for help from the programming community would help diagnose T-embed, creators like Emensta, Popn, Bmorcelli have knowledge.
I'm afraid Lilygo won't help us, I understand their part but this is out of the client's hands.