BruceDevices/firmware #1457: CC1101 8 pin on T embed non CC1101

igniskarya, 2025-07-04:
Hi there  , I have T embed non CC1101.. How to connect RF CC1101  8 pin on it, meanwhile  on pin_arduini.h for non CC1101
#define USE_CC1101_VIA_SPI
#define CC1101_SW1_PIN 43
#define CC1101_SW0_PIN 44
#define CC1101_GDO0_PIN 44
#define CC1101_SS_PIN 43
#define CC1101_MISO_PIN SPI_MISO_PIN
#define CC1101_MOSI_PIN SPI_MOSI_PIN
#define CC1101_SCK_PIN SPI_SCK_PIN

any idea to use that module ( RF CC1101 8 pin) on T embed non CC1101

[image]

Smoking-Dog3, 2025-07-22:
Hi! It all depends on what your t-embed looks like. I have such a board.

[image]

This figure shows the NRF24, but its pinout is completely identical to the CC1101 v2.0
BUT! CC1101 does not work on firmware versions higher than 1.9.1 yet. At least it freezes for me.

igniskarya, 2025-07-24:
> Hi! It all depends on what your t-embed looks like. I have such a board.
> 
> <img alt="Image" width="1254" height="732" src="https://private-user-images.githubusercontent.com/164644693/469262311-1f8118a7-94be-4c68-90af-fd0eff3a99ec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTMzODQzMDIsIm5iZiI6MTc1MzM4NDAwMiwicGF0aCI6Ii8xNjQ2NDQ2OTMvNDY5MjYyMzExLTFmODExOGE3LTk0YmUtNGM2OC05MGFmLWZkMGVmZjNhOTllYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcyNFQxOTA2NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMjM3ZTU2NTJiOGY0NzY3MDc3YTFiM2EzYmRkZjNlYjFjZmM5ZTcyNGQwYjJkMjZkMzg2NTIxN2JkZWY2OGY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.-RIVdrd7DHL5-3LAVCrYwObTp2nCnSRRYu8AU3TqVLI">
> This figure shows the NRF24, but its pinout is completely identical to the CC1101 v2.0 BUT! CC1101 does not work on firmware versions higher than 1.9 yet. At least it freezes for me.

yes ..for NRF24 the wiring diagram like that...eventhough the NRF24 not sensitive to read spectrum ( I tries 3 kinds NRF24 ..even GT24 mini)..

I already discusa with pirata..he recommend using CC1101  8 pin with SW0 & SW1  not used..  I'm trying on it but VCC 3V as a problem ..as you see on those only 1 pin.. coz if VCC used /connected at the same time with NRF24..the status NRF24 & CC1101 will be "Not Found"..so may be for each module must have VCC separately or using switch to activate the module

igniskarya, 2025-08-31:
hi there @Smoking-Dog3 ..have you succed install CC1101 8 pin on t embed non cc1101? could you share the wiring diagram for CC1101 and do you have to change the coding for CC1101?