BruceDevices/firmware #1471: marauder mini schematics needed

maxim1470, 2025-07-11:
I am going to make marauder mini to install bruce on it because it small and i can take it with me.can someone give me schematics how to make marauder mini because in youtube some people doing one schematics and others second schematics. if someone will help me with this i will be very happy.

Liat583, 2025-07-12:
> I am going to make marauder mini to install bruce on it because it small and i can take it with me.can someone give me schematics how to make marauder mini because in youtube some people doing one schematics and others second schematics. if someone will help me with this i will be very happy.

[image]

[image]

Yes, I happen to have it here

Liat583, 2025-07-12:
[image]

[image]

Yes, I happen to have it here

Liat583, 2025-07-12:
[image]

[image]

> [image]
> 
> [image]
> 
> Yes, I happen to have it here