BruceDevices/firmware #1483: Problem with IR in Cardputer v1.1

Mavox-ID, 2025-07-15:
I have a problem with IR, when I power on Cardputer v1.1 to receive a signal, it does not work, there is no signal at all, although another IR sensor receives it calmly, but even with it, for some reason, the signal is not sent, even though I created a signal on it and send it, even from Cardputer v1.1, even from this sensor, IR says "Fail to open file" and when I look at what pin in config>about it says that IR TX: 44 and IR RX: 1

Please help me solve the problem, if anything, I checked the custom firmware with IR checking in the Arduino IDE and everything worked, the problem is in bruce. I also tried everything in IR>config but still the same thing

[image]

[image]

bmorcelli, 2025-07-15:
> I have a problem with IR, when I power on Cardputer v1.1 to receive a signal, it does not work, there is no signal at all,

Cardputer doesn't have IR receiver built in, you nee to connect one into the grove connector, using GPIO 1.

> although another IR sensor receives it calmly, but even with it, for some reason, the signal is not sent, even though I created a signal on it and send it, even from Cardputer v1.1, even from this [...]

To send signals using the m5 IR module you need to change it in the configuration: IR>Config>Ir Tx pin> 2 or Grove Y (or 44 if using the Cardputer led)

You will only "see" the signals being sent with your smartphone camera, so I advise you to check it with the TV-Be-Gone, and check if it is blinking

About the file, you will need to send more information, serial monitor logs and the file itself.

Mavox-ID, 2025-09-22:
I wanted to ask again, what's the problem?