BruceDevices/firmware #1493: Can not cannot to PN532

slooker, 2025-07-17:
I have the PN532 setup based on[ this tutorial](https://www.youtube.com/watch?v=7nNZ8GFfVCg) but the CYD never detects my PN532.  I've included pictures of how I have it hooked up, my solder joints and my errors.  What am I doing wrong here?

[image]

[image]

[image]

Gosheto1234, 2025-07-17:
have you tried playinh with the small switch ? 

[image]

slooker, 2025-07-18:
> i think there was toggle on the switch , also can you check your connections if they are stable on the cyd side ?

The connections seem very stable on the CYD side.  What do you mean a toggle on the switch?

Gosheto1234, 2025-07-18:
> > i think there was toggle on the switch , also can you check your connections if they are stable on the cyd side ?
> 
> The connections seem very stable on the CYD side. What do you mean a toggle on the switch?

There is small switch with 1 and 2 , judging by the picture is set to I2C , you can try using the different configurations , but i am not sure if it will change anything

Gosheto1234, 2025-07-18:
i think I2c is the right but you can try the SPi , and hope to work , not 100% guarenteed