BruceDevices/firmware #1494: [Help] ESP32-S3 with 2.4" TFT Touch (ILI9341) – Custom Build – Black Screen After Flashing Bruce Firmware

GViegas0, 2025-07-17:
Hi everyone, good evening.

I’ve assembled an ESP32-S3-DevKitC-1-N16R8 with a 2.4" TFT Touch display (Cheap Yellow Display style), but so far I’m only getting a black screen — no output at all and no Serial

Screen Driver chip: ILI9341

🔧 Pinout I followed:
| Display Silkscreen | Function     | ESP32-S3 GPIO | Notes                           |
|--------------------|--------------|---------------|---------------------------------|
| T_IRQ              | TOUCH_IRQ    | GPIO 35       | Touch interrupt                 |
| T_DO               | TOUCH_MISO   | GPIO 19       | Touch data (shared with SD)     |
| T_DIN              | TOUCH_MOSI   | GPIO 21       | Touch input (shared with SD)    |
| T_CS               | TOUCH_CS     | GPIO 14       | Touch chip select               |
| T_CLK              | TOUCH_SCK    | GPIO 18       | Shared SPI clock                |
| SDO (MISO)         | MISO         | GPIO 19       | Display/SD data                 |
| LED                | TFT_BL       | GPIO 4        | Backlight                       |
| SCK                | SCK          | GPIO 18       | Common SPI clock                |
| SDI (MOSI)         | MOSI         | GPIO 21       | SPI data to display/touch/SD    |
| DC                 | TFT_DC       | GPIO 16       | Display data/command            |
| RESET              | TFT_RST      | GPIO 17       | Display reset                   |
| CS                 | TFT_CS       | GPIO 5        | Display chip select             |
| GND                | GND          | GND           | Ground                          |
| VCC                | VCC          | 3.3 V         | Power supply                    |

This configuration seems to match what CYD-based boards use.

🧪 What I did:
Flashed the firmware using the Bruce Firmware web flasher

Selected ESP32-S3 as the board

Flash process completed successfully

But… the screen remains black 😓

❓My question:
Can someone help me understand what I might be doing wrong — either in the pinout or in the flashing process?

Should it be manually flashed, and with manual configuration in the repository files?

I’m a noob here  — just trying to get this working!

Thanks a lot in advance!

[image]

sivabala21, 2025-07-19:
I ‘m also facing this issue.

Huzzla101, 2025-07-19:
"This configuration seems to match what CYD-based boards use.

🧪 What I did:
Flashed the firmware using the Bruce Firmware web flasher

Selected ESP32-S3 as the board

Flash process completed successfully"
The issue may be related to the flash your installing but i can not say
this is 100% But maybe you should try to install one of the CYD flashes as
you have done a clean ESP32.S3 installment... And that may not fit in 100%
with your device.... Is it a WROOM BOARD or ?

sivabala21, 2025-07-19:
> "This configuration seems to match what CYD-based boards use.
>
> 🧪 What I did:
> Flashed the firmware using the Bruce Firmware web flasher
>
> Selected ESP32-S3 as the board
>
> Flash process completed [...]

So according to my analysis:

you can install espS3 firmware from flash website
But by default, pins are not configured. The one you install from ESP32-S3 from the website is in headless mode.

We cannot install CYD firmware into the ESP32-S3 model, and unfortunately, only CYD firmware will have the pinouts that we can mock with ESP32-S3
We can go for the Smoothie version, but the display used is different
Finally, the only way is to edit config.ini for the board and compile it

sivabala21, 2025-07-20:
> **In this case, I would suggest that the team could update an ESP32-S3 flasher online with some prompts for drivers and pinouts of typical screens and modules. That should significantly help the community of NOOB DIYers like myself ✌️.**
> 
> Before diving deeper, I should mention that using a custom User_Setup.h configuration in the TFT_eSPI library and the Arduino IDE, I was able to successfully build and run a test sketch on my hardware (ESP32-S3-DevKitC-1-N16R8 with a 2.4" TFT ILI9341 display). This confirmed that my wiring and SPI pin mapping are functional. The hardware pinout details were shared earlier in this thread.
> 
> Now, I’m looking to properly adapt the Bruce Firmware to this setup, keeping full compatibility with the logic , and respecting the structure of the original project.
> 
> **What I’d like to ask the maintainers or advanced users is:**
> 
> Which files should be modified to adapt Bruce to a custom board like the ESP32-S3-DevKitC-1-N16R8? For example, is configPins.h sufficient to configure GPIO mappings for SPI, screen, RF, buttons, etc.?
> 
> Is it possible (or advisable) to select an existing [env] in platformio.ini (like one of the M5Stack or LilyGo entries) and only adjust the pinout and screen driver from there? Or do we need to create a new [env:custom_esp32s3] from scratch?
> 
> Where is the screen driver and rendering logic actually tied to the environment configuration?
> 
> The ultimate goal is to use same approach on this as already used on the CYD boards. A “change this and that” kind of guide would be much appreciated – just a step-by-step on what files to tweak and where to wire things up would really help us DIYers find our way in the firmware jungle😅
> 
> Thanks in advance for any help or pointers from the team or experienced contributors 🙏
> 
> [image]

In Boards ->  ESP General you can find esp32-s3-dev-kit env we just have modify the pins and flash size but the problem is I am facing many issue while recompiling into bin 

there's no clear documentation or tutorial to compile the bin

I am getting errors like :- 

1. Dependences Errors
2. Bootloader Errors

Like you said Maintainers of this project should  give any defaults bin for this s3 Dev kits

GViegas0, 2025-07-20:
> In Boards -> ESP General you can find esp32-s3-dev-kit env we just have modify the pins and flash size but the problem is I am facing many issue while recompiling into bin
>
> there's no clear [...]

Did you make any changes to the ESP-General.ini file?
Or to any of these files \boards\ESP-General:
ESP-General.ini
interface.cpp
pins_arduino.h

I also noticed some warnings (highlighted in yellow) during the build in PlatformIO, but it ends with “Success” anyway…

I’ve already tried using both [env:esp32-s3-devkitc-1-psram] and the one without PSRAM (my ESP32 does have PSRAM), but I still couldn’t get anything to show up on the screen — no backlight or signs of life.

GViegas0, 2025-07-22:
> Share your discord id we will connect.

g_viegas

bettinakreft, 2025-08-16:
> > > [UPDATE] – Build Partially Working but still need some support!!!
> > > Thanks for the tips, [@sivabala21](https://github.com/sivabala21)! 🙏
> > > Here’s a concise update on the changes and trials we’ve made so far:
> > > 1 – Selected [env] smoochiee-board.
> > > 2 – Created a custom pin mapping in boards/smoochiee-board/pins_arduino.h to match our ESP32‑S3 hardware on this custom build:
> > > ```
> > > #define ILI9341_DRIVER 1  // Changed to match screen driver
> > > #define TFT_RST 17
> > > #define TFT_DC 16
> > > #define TFT_MISO 19
> > > #define TFT_MOSI 21
> > > #define TFT_SCLK 18
> > > #define TFT_CS 5
> > > #define TOUCH_CS 14  // SDCARD_CS to ensure SD card works
> > > #define HAS_TOUCH 1
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > 3 – Physical buttons have been disabled by setting SEL_BTN, UP_BTN, DW_BTN, R_BTN, and L_BTN to –1. In boards/smoochiee-board/interface.cpp, to avoid random menu navigation, we modified the code as follows:
> > > ```
> > > void _setup_gpio() {
> > >     if (UP_BTN >= 0) pinMode(UP_BTN, INPUT);
> > >     if (SEL_BTN >= 0) pinMode(SEL_BTN, INPUT);
> > >     if (DW_BTN >= 0) pinMode(DW_BTN, INPUT);
> > >     if (R_BTN >= 0) pinMode(R_BTN, INPUT);
> > >     if (L_BTN >= 0) pinMode(L_BTN, INPUT);
> > >     (...)
> > > }
> > > 
> > > void InputHandler(void) {
> > >     static unsigned long tm = 0;
> > >     if (millis() - tm < 200 && !LongPress) return;
> > >     bool _u = (UP_BTN >= 0) ? digitalRead(UP_BTN) : BTN_ACT;
> > >     bool _d = (DW_BTN >= 0) ? digitalRead(DW_BTN) : BTN_ACT;
> > >     bool _l = (L_BTN  >= 0) ? digitalRead(L_BTN)  : BTN_ACT;
> > >     bool _r = (R_BTN  >= 0) ? digitalRead(R_BTN)  : BTN_ACT;
> > >     bool _s = (SEL_BTN >= 0) ? digitalRead(SEL_BTN) : BTN_ACT;
> > >     (...)
> > > }
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > This update to the InputHandler() logic allows the system to ignore button inputs if the pin is disabled (–1), which solved the random menu navigation issue.
> > > 4 – Successfully flashed the Bruce firmware to our board without build errors.
> > > [HELP NEEDED]
> > > **At this stage, the display colors are still inverted (likely due to an RGB/BGR swap), and touch input isn’t working yet.**
> > > 🔹 **Inverted Colors** We’ve already tried to change, add, or remove several configuration options: TFT_RGB_ORDER bruceConfig.colorInverted = 1; -DTFT_INVERSION_OFF -DTFT_RGB_ORDER=TFT_BGR …but none were able to correct the color inversion.
> > > 🔹 **Touch Input** We attempted to adapt the touchscreen initialization and driver logic from the marauder-touch board (interface.cpp and pins_arduino.h) without success.
> > > Any other suggestions? .
> > > [image] ![Image](https://private-user-images.githubusercontent.com/221426507/471150078-19e2757a-48c2-4461-8556-908c5a422f49.jpg?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTUzMjY1MzgsIm5iZiI6MTc1NTMyNjIzOCwicGF0aCI6Ii8yMjE0MjY1MDcvNDcxMTUwMDc4LTE5ZTI3NTdhLTQ4YzItNDQ2MS04NTU2LTkwOGM1YTQyMmY0OS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgxNlQwNjM3MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNGM5M2ZjODI0MjMzNjAyZDA0YzFhYTA4NGYxNzBkNmMwMmIyNDQzMjg4ZjMzODM3NTcxNjE5ZTJhYzQ1NzRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.3ETO8SEtqa0XNmSiXq-eVWvyyKuA329Rw18UHf8BYco)
> > 
> > 
> > help me sir. I tried to build bruce firmware by changing the tft dc pin to 16 for cyd 2432S028 but failed. here is the error log
> > Linking .pio\build\CYD-2432S028\firmware.elf c:/users/mas dek/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lnet80211 c:/users/mas dek/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lnet80211 c:/users/mas dek/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lnet80211 c:/users/mas dek/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lnet80211 c:/users/mas dek/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lnet80211 collect2.exe: error: ld returned 1 exit status *** [.pio\build\CYD-2432S028\firmware.elf] Error 1
> 
> If your looking to install bruce in esp32s3 with ili9341
> 
> https://github.com/sivabala21/Bruce/tree/main/boards/lord-board
> 
> Build this firmware

no. I want to build bruce with esp32 wroom32.

sivabala21, 2025-08-16:
> no. I want to build bruce with esp32 wroom32.

Then this not branch to speak about the issue

bettinakreft, 2025-08-16:
> Then this not branch to speak about the issue
 
please. Can you help me build firmware bruce CYD-2432S028 for esp32 wroom32 by changing pin tft dc 2 to 16. thanks

closed 2025-08-29 by GViegas0