BruceDevices/firmware #1540: GPS not working on beta version

igniskarya, 2025-08-06:
Hi there..  I have DIY smoochie board ...on the previous version ( 1.10.1 & 1.10.2) GPS working well..but when I flasher into the beta version GPS ..status just WAITING...

Same problem with Marauder mini on beta version status just waiting.

Any solution or there are some changes on coding with GPS module??

archiepa, 2025-08-10:
Check if baud-rate 115200

igniskarya, 2025-08-14:
I have already re-check for all of baud rates options. Test for waiting almost 15 minutes ( for each baud rates option ..GPS blinking in 2-3 minutes) but status just waiting...

but it I re-installed to the latest in 2 minutes GPS will be locking....can someone share solving this issue??

filon543, 2025-09-09:
After Release 1.10.2, GPS doesn't work for me on all subsequent versions (pending)???