BruceDevices/firmware #1564: Bruce 1.10.2 Freezing on CC1101 in Multiple Menus

absinthe5324, 2025-08-16:
I am experiencing an issue with Bruce version 1.10.2 on the CC1101 module. The application frequently freezes and becomes unresponsive in several menus, requiring a reset to recover functionality.

Details:

The freezing occurs in various menus, including:
FM Scan
RF Spectrum
When navigating to these sections, the application hangs, and the only way to regain control is by pressing the reset button.

Steps to Reproduce:

Open the Bruce application.
Navigate to the FM Scan menu.
Observe the application freezing.
Alternatively, navigate to the RF Spectrum menu to replicate the issue.
Expected Behavior:

The application should function smoothly without freezing in any menu.

Actual Behavior:

The application becomes unresponsive, necessitating a reset.

Please investigate this issue at your earliest convenience. Thank you!

kulasa07, 2025-08-17:
I encountered the same problem 😓

bmorcelli, 2025-08-21:
Does it happen only if you are connected to you computer with the USB Chord? 

I noticed that it happens to me only in this condition, no other...

And it happens because your computer is flooding the Serial Interface with invalid "Serial cmd" commands, so it glitches.. but stops when I unplug the USB cable.

absinthe5324, 2025-08-25:
> Does it happen only if you are connected to you computer with the USB Chord?
>
> I noticed that it happens to me only in this condition, no other...
>
> And it happens because your computer is flooding [...]

 I have not specifically tested whether the freezing occurs only when connected to my computer via the USB cable. I will definitely check this out to see if the issue is related to the USB connection.