BruceDevices/firmware #1591: t-embed battery drops

davidezambon06, 2025-08-28:
hi guys, while i was using the nrf jammer function on my t-embed, that is running bruce firmware, the battery instantly dropped fron half % to 10%, and now it wan’t charge over 70%, what i have to do?

PixelGames987, 2025-09-08:
I am also facing the same exact problem, I don't know it it's a problem with the t-embed or with the Bruce firmware though.

davidezambon06, 2025-09-08:
maybe chance the battery will solve the problem?

smoochiee, 2025-09-08:
its normal its power hungry that gauge cannot handle..maybe turning off the pmic read % would help

davidezambon06, 2025-09-08:
whats pmic?

PixelGames987, 2025-09-08:
I have tried a few different batteries, both li-ion and li-po and it gave me the same result. So it's unlikely that the battery is the problem.

smoochiee, 2025-09-08:
disable battery gauge while using this nrf specially the high power once...it pulling much current that gauge cant handle

Gosheto1234, 2025-09-09:
> disable battery gauge while using this nrf specially the high power once...it pulling much current that gauge cant handle

how do you do that ? sorry i am new to the T embed and have not much experience , can you explain how you disable battery gauge ?

davidezambon06, 2025-09-15:
i dont know how too