BruceDevices/firmware #1607: Smoochie board battery gauge

SRX-YT, 2025-09-05:
so i build smoochie bruce pcb v2, im tryed official firmware from the web flasher, and buttons acts like there is no pull up resistors, so i have downloaded repo and in interface.cpp i changed INPUT to INPUT_PULLUP and now its worked fine but i cant see battery gauge, why?

Yag00000, 2025-09-10:
in Smoochie Board DYC w GPIO ist Esc key?

GViegas0, 2025-09-26:
Hi there,
I had exactly the same problem and it was only solved when I moved GPIO0 to another pin (42). I used INPUT_PULLUP and kept the external pull-up resistors to VCC on the button wires.
Did you find how to enable the battery indicator on the Smoochie?

Thanks

bezdarnosti-yt, 2025-09-27:
> Hi there,
> I had exactly the same problem and it was only solved when I moved GPIO0 to another pin (42). I used INPUT_PULLUP and kept the external pull-up resistors to VCC on the button wires.
> Did you find how to enable the battery indicator on the Smoochie?
> 
> Thanks
> 

Let me know if you found some!;)