BruceDevices/firmware #1609: Smoochie Board DYC

Yag00000, 2025-09-06:
I am working in a DYC smoochie board w my S3 NR16 , but how can the color in the RGB led , always turn white !! , display 1.9 inchs works fine,  only fix buttons menu tactil !!

igniskarya, 2025-09-06:
are you using RGB LED or smart RGB LED...?

An RGB LED is a basic component of Red, Green, and Blue diodes that produce different colors, while a WS2812B is a specific type of addressable, "smart" RGB LED that has an integrated controller and driver chip. 

The key difference is that a WS2812B allows each individual LED on a strip to be controlled independently for complex animations, whereas a standard RGB LED typically requires multiple components and more complex wiring to achieve similar, simpler effects.

igniskarya, 2025-09-06:
coz on smoochie board

#define LED_TYPE WS2812B ( Type of  RGB LED)

Yag00000, 2025-09-10:
what GPIO its Esc key ?

Yag00000, 2025-09-16:
[image]