BruceDevices/firmware #1616: Cardputer doesn't play sound on boot

JoplayXYZ, 2025-09-09:
**Describe the bug:**
When starting the bruce firmware there is no boot sound

**Steps To Reproduce:**
Start any recent version of the firmware

- Device used: M5stack Cardputer v1.1

**Expected behavior:**
On booting the firmware the integrated speaker module should play a short tone such as on the m5stack stick c plus 2

**Actual behavior:**
When starting the bruce firmware on a m5stack cardputer v1.1 there is no boot sound 

**Additional context / logs:**
(No unusual activity)

SDCARD mounted successfully
[  2065][I][config.cpp:430] fromFile(): Using config from file
{
  "B0:81:84:95:23:30": {
    "CC1101_Pins": {
      "sck": 40,
      "miso": 39,
      "mosi": 14,
      "cs": 1,
      "io0": 2,
      "io2": -1
    },
    "NRF24_Pins": {
      "sck": 40,
      "miso": 39,
      "mosi": 14,
      "cs": 1,
      "io0": 2,
      "io2": -1
    },
    "SDCard_Pins": {
      "sck": 40,
      "miso": 39,
      "mosi": 14,
      "cs": 12,
      "io0": -1,
      "io2": -1
    },
    "W5500_Pins": {
      "sck": 40,
      "miso": 39,
      "mosi": 14,
      "cs": 1,
      "io0": 2,
      "io2": -1
    }
  }
}[  2114][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 160
PNG Loaded in 105ms
[  2698][E][vfs_api.cpp:105] open(): /sd/shortcuts.json does not exist, no permits for creation
[  2706][E][mykeyboard.cpp:148] checkShortcutPress(): Shortcuts Config file not found. Using default values

bmorcelli, 2025-09-09:
It only plays if you have a "boot.wav" at the root of SdCard or LittleFS.

JoplayXYZ, 2025-09-09:
Why doesn't it have a default boot sound like on the M5stick c plus 2