BruceDevices/firmware #1649: Ethernet CRASH/FREEZE

naofumi39538, 2025-09-16:
**Describe the bug:**
The Ethernet new feature from newest version, when i click it, my device crash/freeze, buttons are useless except the shut down and turn it on again to fix the freezing screen

**Expected behavior:**
It wont crash, i dont know the inside of the ethernet feature, if it crash, maybe switch it into "w5500 chip not found" just like the nrf24 when nrf24 not attached to the device

**Actual behavior:**
When i click the Ethernet, it crashed/freeze so i just shut down and reopen it again, even if i try to click the Ethernet again, it still freeze my device

**device used:**
M5stickC plus 2

**Version:**
1.11

This is not the only issue i encountered, theres more than this

emericklaw, 2025-09-16:
I am debugging this now.

fernandobarros01, 2025-10-06:
> Working on this now For Smoochie DIY

Theres any update for use ethernet card in Smoochie DIY?