BruceDevices/firmware #1660: Can't find any firmwere for ESP32 S3-WROOM-1 Dual Type-C USB N16R8 Micropython Board

zoominzoo960-byte, 2025-09-20:
Im using ESP32 S3-WROOM-1 Dual Type-C USB N16R8 Micropython Board

But cant find any compatible bruce firmware for it i tried many but no luck, 
And i also tried smoochi firmwere but it does nothing,just my board led glows bright white, and my oled display also got broken while making is there is any firmwere that i can flash in this board and it also has a wifi acess point for web view,

Tristan-Eternis, 2025-09-20:
if you know how to build from source, you can use these config files (if your screen is an ILI9341) https://github.com/sivabala21/Bruce/tree/main/boards/lord-board

QuantumKraken0, 2025-09-24:
as long as the screen has the ILI9341 controller, it should work fine if you check the pin mappings in the pins_arduino.h inside the lord-board folder