BruceDevices/firmware #1671: Compilation failed due to NimBLE-Arduino incompatibility, ESP PN532Killer library version problem and CYD -DEVICE_NAME conflict.

fsadak, 2025-09-24:
Hello,

Issue 1:

I am trying to compile a project that uses the ESP-PN532BLE library and I'm encountering compilation errors. It seems to be an incompatibility issue with the NimBLE-Arduino library.

Steps to Reproduce:

Clone the project from https://github.com/pr3y/Bruce.
Open the project with PlatformIO.
Try to compile the project using the CYD-2USB environment.
Expected Behavior:

The project should compile without errors.

Actual Behavior:

The compilation fails with the following error log:

.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp: In destructor 'PN532_BLE::~PN532_BLE()':
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:16:23: error: 'isInitialized' is not a member of 'NimBLEDevice'
     if (NimBLEDevice::isInitialized())
                       ^~~~~~~~~~~~~
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp: At global scope:
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:24:1: error: expected class-name before '{' token
 {
 ^
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp: In member function 'void scanCallbacks::onResult(const NimBLEAdvertisedDevice*)':
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:27:39: error: passing 'const NimBLEAdvertisedDevice' as 'this' argument discards qualifiers [-fpermissive]
         if (advertisedDevice->getName().find("PN532") != std::string::npos &&
                                       ^
In file included from .pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEScan.h:20,
                 from .pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEDevice.h:22,
                 from .pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.h:12,
                 from .pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:9:
.pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEAdvertisedDevice.h:74:21: note:   in call to 'std::__cxx11::string NimBLEAdvertisedDevice::getName()'
     std::string     getName();
                     ^~~~~~~
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:28:39: error: passing 'const NimBLEAdvertisedDevice' as 'this' argument discards qualifiers [-fpermissive]
             advertisedDevice->getName().find("BLE") != std::string::npos)
                                       ^
In file included from .pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEScan.h:20,
                 from .pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEDevice.h:22,
                 from .pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.h:12,
                 from .pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:9:
.pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEAdvertisedDevice.h:74:21: note:   in call to 'std::__cxx11::string NimBLEAdvertisedDevice::getName()'
     std::string     getName();
                     ^~~~~~~
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp: In member function 'bool PN532_BLE::searchForDevice()':
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:119:12: error: 'class NimBLEScan' has no member named 'setScanCallbacks'
     pScan->setScanCallbacks(new scanCallbacks(), false);
            ^~~~~~~~~~~~~~~~
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:123:67: error: no matching function for call to 'NimBLEScan::getResults(int, bool)'
     NimBLEScanResults foundDevices = pScan->getResults(5000, false); // 5 seconds in milliseconds
                                                                   ^
In file included from .pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEDevice.h:22,
                 from .pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.h:12,
                 from .pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:9:
.pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEScan.h:78:25: note: candidate: 'NimBLEScanResults NimBLEScan::getResults()'
     NimBLEScanResults   getResults();
                         ^~~~~~~~~~
.pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEScan.h:78:25: note:   candidate expects 0 arguments, 2 provided
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:128:82: error: cannot convert 'NimBLEAdvertisedDevice' to 'const NimBLEAdvertisedDevice*' in initialization
         const NimBLEAdvertisedDevice *advertisedDevice = foundDevices.getDevice(i);
                                                                                  ^
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:129:39: error: passing 'const NimBLEAdvertisedDevice' as 'this' argument discards qualifiers [-fpermissive]
         if (advertisedDevice->getName().find("PN532") != std::string::npos &&
                                       ^
In file included from .pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEScan.h:20,
                 from .pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEDevice.h:22,
[... 211 lines cut ...]
     std::string     getName();
                     ^~~~~~~
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp: In member function 'bool PN532_BLE::searchForDevice()':
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:119:12: error: 'class NimBLEScan' has no member named 'setScanCallbacks'
     pScan->setScanCallbacks(new scanCallbacks(), false);
            ^~~~~~~~~~~~~~~~
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:123:67: error: no matching function for call to 'NimBLEScan::getResults(int, bool)'
     NimBLEScanResults foundDevices = pScan->getResults(5000, false); // 5 seconds in milliseconds
                                                                   ^
In file included from .pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEDevice.h:22,
                 from .pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.h:12,
                 from .pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:9:
.pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEScan.h:78:25: note: candidate: 'NimBLEScanResults NimBLEScan::getResults()'
     NimBLEScanResults   getResults();
                         ^~~~~~~~~~
.pio/libdeps/CYD-2USB/NimBLE-Arduino/src/NimBLEScan.h:78:25: note:   candidate expects 0 arguments, 2 provided
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:128:82: error: cannot convert 'NimBLEAdvertisedDevice' to 'const NimBLEAdvertisedDevice*' in initialization
         const NimBLEAdvertisedDevice *advertisedDevice = foundDevices.getDevice(i);
                                                                                  ^
.pio/libdeps/CYD-2USB/ESP PN32BLE/src/pn532_ble.cpp:129:39: error: passing 'const NimBLEAdvertisedDevice' as 'this' argument discards qualifiers [-fpermissive]
         if (advertisedDevice->getName().find("PN532") != std::string::npos &&

Issue 2:

in boards\CYD-2432S028\CYD-2432S028.ini file [CYD_base] section has -DDEVICE_NAME='"CYD-2432S028"' macro. Also there is same defination in [env:CYD-2432S028] and in [env:CYD-2USB] section -DDEVICE_NAME='"CYD-2USB"' also defined. These defines issues a conflict and i solved this problem delete the defines -DDEVICE_NAME='"CYD-2432S028"' from  [CYD_base] and [env:CYD-2432S028].

fsadak, 2025-09-28:
> > ของฉันทำงานได้ดีแต่จอ LCD เป็นสีขาว (ระบบสัมผัสใช้งานได้)
> 
> Can you compile it? Can I see your setup? I have the same error as the poster.

No, i cannot compile already and i gave up.

wh1ter0z, 2025-09-28:
> > > ของฉันทำงานได้ดีแต่จอ LCD เป็นสีขาว (ระบบสัมผัสใช้งานได้)
> > 
> > 
> > Can you compile it? Can I see your setup? I have the same error as the poster.
> 
> No, i cannot compile already and i gave up.

Hi 

I have been compiled and now works good but no advertising at all if you dont want ble yet i can give you files to replace it for compile 

I fixed the white screen

 (set a user setup in tft_espi->user_setups->mycustom.h then in your folder chosen like cyd-blablabla you have to iclude it ) 

on my diy esp32 wroom and 3.5 inch tft ili9488 (i build it like cyd model) 

Tell me if you want 

(Telegram: @thereisnomind)

wh1ter0z, 2025-09-28:
> > ของฉันทำงานได้ดีแต่จอ LCD เป็นสีขาว (ระบบสัมผัสใช้งานได้)
> 
> Can you compile it? Can I see your setup? I have the same error as the poster.

I can help you so if you want dm me 

I compiled but Ble is not works (advertising not working)
Just find Ble and Bluetooth (i have to fix it ) 

If you want i can send that fixed library for compile to pass it

closed 2025-12-10 by bmorcelli

YOUNESSBEN18, 2026-03-29:
Can you give me compatible version of NimBle_Arduino compatible with marauder